Xinyuan-LilyGO/T-FPGA #9: Unable to program FPGA on Linux Mint

ChrisJones79, 2024-09-21:
I read through the README and was able to upload the 'turn_on_power' project to the ESP32. I see the LEDs behaving appropriately and saw that the terminal in platformio reported a successful programming. Now, I have opened the Gowin IDE and all has compiled (it was already) and the programmer opens, as expected. 

Here is the log from the programmer, while trying to upload to the FPGA. I do have the USB-C connected to the FPGA-JTAG port, too.

```
Info:	Cable found:  Gowin USB Cable(GWU2X)/0/17/null (USB location:17)
Info:	Cost 0.02 second(s)
:	 
Info:	Cable found:  Gowin USB Cable(GWU2X)/0/17/null (USB location:17)
Info:	Cost 0.01 second(s)
:	 
Info:	Target Cable: Gowin USB Cable(GWU2X)/0/17/null@1.33 MHz
Info:	Target Device: GW1NSR-4C(0x0100981B)
Info:	Operation "SRAM Program" for device#1...
Error:	Cable open failed.
Info:	Cost 0.0 second(s)
:	 
Info:	Target Cable: Gowin USB Cable(GWU2X)/0/17/null@1.33 MHz
Info:	Target Device: GW1NSR-4C(0x0100981B)
Info:	Operation "SRAM Program" for device#1...
Error:	Cable open failed.
Info:	Cost 0.0 second(s)
:
Info:	Scanning!
Info:	Target Cable: Gowin USB Cable(GWU2X)/0/17/null@1.33 MHz
Error:	Cable open failed.
Error:	No Gowin devices found!
Info:	Cost 0.02 second(s) 
```

Also:
``` 
$ lsusb
Bus 001 Device 037: ID 33aa:0120 GOWINSEMI GWU2X
```

and plugging in while udev monitoring:

``` 
$ udevadm monitor
KERNEL[675634.337777] add      /devices/pci0000:00/0000:00:14.0/usb1/1-2 (usb)
KERNEL[675634.339741] add      /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0 (usb)
KERNEL[675634.339944] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-2 (usb)
UDEV  [675634.376606] add      /devices/pci0000:00/0000:00:14.0/usb1/1-2 (usb)
UDEV  [675634.378558] add      /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0 (usb)
UDEV  [675634.383025] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-2 (usb)

```

I don't have any `udev` rules added for the T-FPGA, and I didn't see anywhere in the repo where the rules would be.

Here's a screenshot from the programmer:
[image]

Any help? Thanks.

46cv8, 2025-11-25:
Did you ever figure this out I'm running into a similar error ....

ChrisJones79, 2025-11-27:
@lewisxhe @LilyGO is adding the udev rules part of the Gowin installation process? Could this be a case of several entities assuming the others would add the udev rules? platformIO, Gowin, and LilyGO? Also, thank you for updating with a solution, @46cv8

lewisxhe, 2025-11-28:
Sorry, I only saw this after you tagged me. I didn't maintain this repository before, so I didn't follow this issue. I have no experience using this on Linux; all my tests are based on Windows.

Have you checked the gowinsemi documentation center?

https://www.gowinsemi.com/en/support/database/14/